image-et-son/p600fw #110: Mod wheel LFO depth at low values may still be too much

matrix12x, 2022-02-02:
I was testing the mod-wheel in the latest alpha and while I love the range for PWM/filter cutoff/ and VCA... 
VCO seems like there not enough playable range on the mod wheel before it becomes wildly unusable for simple vibrato. I would never go anywhere near that max for a simple vibrato.

For me, I used to play with the mod wheel with VCO on the mod wheel range on min. I effectively have a useable range of values 0 thru 15 now before it becomes "too much."  If I max out the lfo frequency, then the mod wheel range on the VCO is awesome, but not so great for subtle vibrato for soloing.

Can we expand that range somewhat? Or add back the "min" setting option?

el-folie, 2022-02-02:
I have to admit I have no problem of using the mod wheel for subtle modulation by just nudging it up a bit by muscle memory. To me the current scaling is excellent. But every use case can be different and so I totally see your point. Also, if full range VCO mod depth lies in the future, maybe we really may need back a lo/mid/hi switch for VCO mod depth on the mod wheel.

matrix12x, 2022-02-02:
I think my modwheel is scratchy/dirty around the lower values also.  I may need to clean it.

image-et-son, 2022-02-03:
@matrix12x : the range is full, I think? Do you need a softer onset or would you like to use the full range of the mod wheel with a (strongly) reduced maximum strength?

matrix12x, 2022-02-03:
For how I use it, only values 00-20 or so are currently useable for me.  I think that means I would like an option with a much lower max strength.

image-et-son, 2022-02-03:
I am tempted to just cut the maximum by 2. I really don't think that anyone would miss that. If you want to go crazy you can still use the LFO Amount. I'd  try to keep the max VCA depth, though.

image-et-son, 2022-02-03:
Check it out in alpha 13.6. What do you think?

matrix12x, 2022-02-07:
Alpha 14:
I really like "touch" and "soft." they both work really well for my use case.
I like how I can subtly bring in vibrato (modulating VCO A/B using the lfo) across the entire mod wheel range.

I'll check out the performance of the other controls. But so far I tweaked cutoff and main volume and they seemed responsive. (poly mode - one note at a time pattern playing lead lines)

image-et-son, 2022-02-08:
OK, now the combination of features: for "high" and "full" I have made the mod wheel curve more concave, e.g. slower onset. In the case of "full" this corresponds to exactly the same curve as the earlier version with still very smooth action and full modulation at max. I will post this as alpha 14 once complete. I am doing some checks before...

el-folie, 2022-02-08:
Looking forward and thx in advance for the new/old "smooth onset full range" setting. 👍

matrix12x, 2022-02-27:
The set of ranges in the current Alpha 14 (released today) works well for me.

closed 2022-02-28 by image-et-son