image-et-son/p600fw #111: Bug - Alpha 14- in SCI panel mode "default patch"  has a drive level of 0 (but shows 61)

matrix12x, 2022-02-07:
I think I found a bug, please confirm:
Alpha 14

in SCI panel mode I press "from tape" + "preset" I get no sound. (Initially I thought this was mix being all the way to VCOB again, but its not)

I checked various parameters and drive seems to actually be at zero. I say actually, because if I leave it in "to tape" so I cans values, and on Drive, when I press "from tape" + "preset" Drive comes up as "61" but it is still dead silent. then when I turn the speed knob, drive suddenly works and I now have sound.

Please note in GliGli mode, there is no such similar issue when I press "from tape" + "preset" 

Please confirm this issue, I often wonder if random bugs are just my synth.

el-folie, 2022-02-07:
I just wanted to reflash to RC14 by syx in order to test for the bug you described and after reflashing again the tuning was lost. At a second try to reflash by syx the running square stopped while flashing, no S for success, after that no reflashing possible anymore, LCD stayed on "U". Then reflashed by hex. Will test now...

el-folie, 2022-02-07:
This time, after reflashing to RC14 and still in GliGli mode I pressed a key, it made sound, then switched to SCI mode, and it also made sound instantly. There seems to be no rhyme or reason to when the osc volumes are down or not.

Then tested the bug you found. I can confirm no sound on INIT patch "From Tape plus Preset twice". Drive parameter on LCD was 00, raising its value immediately mada the oscs sound.

el-folie, 2022-02-08:
Additional notice for imogen: 
The panel memory wipe/lost tuning this time happened from RC14 to reflashing to RC14 by syx. So RC14 still exhibits this behaviour.

el-folie, 2022-02-08:
Next interesting thing:
When the tuning/note allocation table is lost in panel mode, by using the INIT patch (and drive value up) in preset mode, notes are tracking the keyboard as they should. So the problem of "lost note allocation" must be "panel mode saving" related.

matrix12x, 2022-02-08:
When I was using SCI panel mode, I just had the weird tuning issue. I recalled a patch in preset mode and tuning was fine. I did not reflash.

el-folie, 2022-02-08:
When I turned on my p600 today, the tuning in panel mode was lost again. When I switched the machine off yesterday night it still worked. Then today I tried to reflash a few times with older alphas by hex to get tuning back - didn´t work! That´s a first... I needed to go back to 2.1rc3 as hex to get tuning back. Weird! Then I could flash my way back up to RC14 with intact tuning. The "lost tuning" problem is a really spooky as it happens so random. JUst an idea - maybe the Teensy memory is dying from too much reflashing?

matrix12x, 2022-02-08:
What happens if you hit tune with the list tuning?  
when these tuning corruptions happen I always assumed that where the tuning was stored was being overwritten partially by something else

matrix12x, 2022-02-08:
Program wise, I wonder what actually gets scrambled when this happens. It's almost like the assigner gets bad info sent to it.

I've only had this happen once so far. But I really don't use SCI mode except for testing the Alphas. In my case it happened when I was in SCI panel mode, and had just pressed "from tape" + "preset" then I adjusted "drive", then when I turned "mix" to add some VCO B and toggled the waveform switches for VCO B was when I noticed it.

el-folie, 2022-02-08:
Hi, yes panel/live/manual mode as opposed to preset mode.

I first had the suspicion that the SCI panel mode may be the cause why and when the weird lost tuning thing happens, but it seems to be happening random and not only in SCI mode. Still I could imagine that the introduction of SCI panel mode and/or different handling of the memory available on the Teensy board may be somehow involved. I never experienced this bug before we got the SCI mode back.

I tried to provoke/confirm the tuning loss by your procedure above, but the tuning stayed intact. It´s really hard to find the exact cause.

image-et-son, 2022-02-14:
I'll fix the Drive=0 issue here. "Tuning loss" or rather "zero load" is tracked in the other issue #67

matrix12x, 2022-02-14:
Regarding the Zero Drive (SCI panel mode) when initiating a default patch, it is still there under the current Alpha. Just verified.

el-folie, 2022-02-14:
RC14_c2: Just verified - drive on 00 on basic patch initialization.

matrix12x, 2022-02-27:
I confirm that the current Alpha 14 seems to correct this issue.

closed 2022-02-28 by image-et-son