image-et-son/p600fw #112: Bipolar pot values in display change to non-bipolar values when a switch is changed

image-et-son, 2022-02-21:
In live mode or preset panel mode when the display shows the value of the last touched pot changing any switch change the displayed value to its 0..99 range value, which is a bug if the displayed value should be bipolar -50 to 50.

el-folie, 2022-02-21:
Confirmed, also if the last knob moved was osc freq, after a switch it´s changed to its 0...99 value.

matrix12x, 2022-02-27:
The current alpha seems to fix this. I tested by turning on "to tape" so I can see values and then tweaking a bipolar pot (like env amount). Then I switched every switch. The display did not go to 00 or 99, it remained on the current value for the last tweaked bipolar pot.  I tried in preset and manual mode.

closed 2022-02-28 by image-et-son