image-et-son/p600fw #133: Knob pick up problems again

RDPRDP, 2022-03-06:
I thought I had all issues revolved, but tonight I had problems picking up OCS A Gain/Vol.

matrix12x, 2022-03-06:
Did you have the issue with OSC A volume in GLIGLI panel mode or SCI panel mode? 
"From tape" + 6 switches between these modes (pages 47-48 of new manual)

RDPRDP, 2022-03-07:
On GLIGLI, which is what I normally use. I've been able to find a single patch that gives me this issue. I can go back and forth between this patch and another, and while the other works, this specific patch will always give the same issue. I'd upload the SYSEX, but I'm having trouble finding something to send a SYSEX REQUEST command to the synth so I can record the single SYSEX (not a problem) and attach it for you to try if you like.

matrix12x, 2022-03-07:
If you press "from tape" + Record, "Preset" will be flashing. Then press the patch number (e.g., 05 or 11 or whatever), it dumps the patch. I use the program C6 on my mac to send and receive sysex.

RDPRDP, 2022-03-07:
I switched my interface over to my PC to try the dump. Hopefully this is the right patch. I'll move back to my OSX setup and send it back.

Note: Thansk for letitng me know single dump is available, I was using the SYSEX solution here! [https://gearspace.com/board/showpost.php?p=15778152&postcount=3116](url)
[OSCGAIN.zip](https://github.com/image-et-son/p600fw/files/8194182/OSCGAIN.zip)
.

matrix12x, 2022-03-07:
Before loading into my P600 I was going to look at the parameters of this file (above).

@image-et-son for some reason the SYSEX parameter converter isn't reading this file right (attached above). It says there is no F7 at the end, but there is an F7 at the end.

"Non SysEx structure: proper F7 closure missing or byte sequence incomplete. Stopped at input byte:  170"

RDPRDP, 2022-03-07:
Just to note, please ensure to use V2 now updated in my initial attachment post. The first wasn't the right patch.

matrix12x, 2022-03-07:
Thanks. Just did that. V2 works in the patch parameter converter. I was trying to look at the parameter values before loading it into my P600 for testing.

Might it be a corrupt patch? If it is the only patch that has the bug, it may not be an OS bug. Either way I'll check it out.

RDPRDP, 2022-03-07:
V1 was probably some user error as I had quickly jumped over to MIDI-OX on my PC and just wildly tried to obtain a result, without really knowing the full intricacies of the software. For V2 I reverted back to my Mac, and used your single dump command from the P600 panel. Now, if there is possibility for an error, maybe it's in the situation where a SYSEX REQUEST is sent, as that is how I accomplished V1. The command was: F0 00 61 16 02 03 F7 (Patch located in memory 03).

matrix12x, 2022-03-07:
I confirm for RDP's patch above, I could not get pick up to on OSC A volume, only if I had "to tape" already on when I tweaked the knob. If I loaded the patch and had the "to tape" light off, the knob picked up fine and chnages to OSC A volume worked as expected (same patch). 

If I started with the "to Tape" led on, pickup would not happen for VCO A volume, but if I turned "to tape" off and on again, then it picked up fine and then that patch would never have that issue again. I was able to repeat this consistently.

I'm not sure if it is a patch bug, or a coding bug. @RDPRDP was this patch made in a prior Alpha? (not then current Alpha 14), if so, that may explain the issue.

RDPRDP, 2022-03-08:
Glad to see you were able to reproduce my issue!

I've read your notes a few times, and am possibly confused on how things work. For me as well, if I do not have "TO TAPE" selected I can change the values without issue, but my thinking is this works as pickup mode is not engaged, and the value just jumps to the current value, in the same manner the synth operated before we ever got into trying to achieve a pick-up mode? In other words, if "TO TAPE" is not selected, you should never have an issue, as the routine is not engaged, trying to resolve your position against the memory position.

I also tried your solution. Turn "TO TAPE" on, then off (without changing the pot!!!!), then on again, but for me, I am unable to get the pickup to work, the outcome is the same.

fun fun fun

RDPRDP, 2022-03-08:
I forgot to answer: I'm not sure when I created this patch, but pretty sure it was a prior Alpha. Probably quite a versions old. Too bad patch SYSEX doesn't capture the HOST OS.

image-et-son, 2022-03-09:
> I forgot to answer: I'm not sure when I created this patch, but pretty sure it was a prior Alpha. Probably quite a versions old. Too bad patch SYSEX doesn't capture the HOST OS.

The storage version should be sufficient in terms of capturing context info. When you use the script to see the values stored in the patch then this is what is used inside the P600 no matter what the OS state was it was created in. No other/hidden information that affects the behaviour. 

I will try to reproduce the bug. I will first look at the values of vol A and vol B in the patch. In my very first attempt of pick-me-up there were pots I couldn't pick up because the value was too large for the accuracy of the pot scanning. While I fixed that it might be a similar constellation.