image-et-son/p600fw #134: Know pickup interrupts editing

RDPRDP, 2022-03-06:
If you have a knob in pickup mode, and don't actually seek the current value the indicator on the led will remain either pointing left or right. This is not really a problem, but if you then decide (without picking up the knob) to try and edit a parameter on the membrane panel by selecting a number, the pickup graphics interferes with the display, and you cannot do anything, except exit the mode (press TO TAPE), then enter again (press TO TAPE).

image-et-son, 2022-03-09:
Note: the problem is not related to pick-up mode symbols. It also happens for displayed pot values, e.g. after pressing a button to select a menu parameter the last pot value is shown again.

image-et-son, 2022-03-26:
Please check in version 15. Should be OK now.

RDPRDP, 2022-03-27:
Yes, seems to be perfect now!

closed 2022-03-27 by image-et-son