image-et-son/p600fw #135: Feature request: LFO synced divisor, and phase offset

RDPRDP, 2022-03-06:
When sync'ing to MIDI, it would be nice to control the LFO so:

1. You can set the start phase on trigger/reset of the LFO shape.
2. The speed dial works on divisors of the MIDI clock for the LFO.  Maybe also retain the current RATE as well, though an option switch?

image-et-son, 2022-03-09:
Hi, when you sync clock to external MIDI then the data/speed dials already sets the MIDI clock divider (instead of speed), so it also affects the LFO if the LFO is set to be sync'ed to seq/arp clock. Have you tried that? See manual :-)

BTW: at the moment LFO sync is only active if arp and/or seq is running. Is there a case for having LFO sync'ed to external MIDI also if there is no arp or seq running?

The LFO phase is be a new feature, but as a feature request it is pretty clear.

image-et-son, 2022-04-05:
@RDPRDP: Hi, have you had a chance to look at my comments and questions relating to your feature request?

RDPRDP, 2022-04-07:
@image-et-son I believe we completed the conversation here: https://gearspace.com/board/showpost.php?p=15888843&postcount=3468 . 

(Just to clarify, I would be very happy to have a solution to play a midi sequence from my computer, with or without MIDI clock running, or on the P600 keyboard directly, but without the ARP or Sequencer necessarily running. This permits a guaranteed response from the LFO in regards to the trigger and envelopes. A great usage of this would be if we ever move to more expanded LFO shapes as per the Martix12x builds. Even now with the stock LFO shapes it is still useful. For instance, you can have filter action (LFO to FLITER) that is repeatable (you know what the response will be).

Originally Posted by RDP [➡️](https://gearspace.com/board/electronic-music-instruments-and-electronic-music-production/845271-prophet-600-users-have-upgrade-post15888742.html#post15888742)
Thanks everyone for you help. Two things I am seeing.

1. Does the LFO sync work only with the sequencer/appregiator? I was trying to use it with MIDI notes and locally played note triggers.

2. Setting sync to INT when using appregiator makes it work, I can now hear the differences the LFO sync function offers. However, can this not also work with notes (MIDI or local)? Unless I am confusing functionality In the [Access Virus](https://gearspace.com/gear/access-music/virus-a) they described it as Envelope Mode, but I see the trigger was done on each note. In practice I would like to use the LFO in a repeatable fashion, therefore to sync note by note, or by a bar division, or by tempo division is necessary.

thanks,
Robert
I guess these are different functions. LFO Sync in the new firmware means sync'ing the LFO to the beats of the arp/seq. What you describe is more akin to setting LFO to retrigger with new notes (with variants, e.g. typically only after all notes have been released).

The sync implemented now is meant to provide complex but rhythmic modulation patterns over arpeggiator and sequencer.

image-et-son, 2022-04-07:
OK, understood. 

If this feature request gets to be selected I would need to think if the clock can always run and how/where a new phase parameter can be implemented.

image-et-son, 2022-09-21:
Hi @RDPRDP: I am in the process of sweeping up all the ideas and requests. After this discussion I had already introduced the LFO Sync option "key", where the LFO is retriggered with a new note (when no note is held). This is a feature of v2022. That covers half of the story, I guess. 

So I figure there are three residual ideas, right? 
- Parameter for setting the phase offset
- Having the internal clock running also without arp/seq running so that LFO can be sync'ed to it
- Combination of "on key" LFO trigger with LFO sync to clock?

RDPRDP, 2022-09-22:
Hello!
Yes, that is all right. The only modification might be (maybe this variable
is covered):

   - Having the internal clock running also without arp/seq running so that
   LFO can be sync'ed to it

change to:

   - Having the *internal/external/midi* clock running also without arp/seq
   running so that LFO can be sync'ed to it

And maybe a CC or another source to modulate the phase offset! Oh.......
that sounds interesting! Beyond scope I think.....

And having the extra Martix12x LFO shapes back would be a bonus!

thanks again!
Robert