image-et-son/p600fw #136: Patch names

RDPRDP, 2022-03-08:
To facilitate better patch management in a library, maybe patches can have names?

matrix12x, 2022-03-08:
The current sysex format that the P600 uses enables the use of patch names. So if someone made a librarian, we already can name the patches. We just need someone to make a librarian.

image-et-son, 2022-03-09:
The new firmware includes a storage part intended for patch naming (16 8-bit values). At this stage the firmware does not offer any features to edit a patch name or to display the name on the P600. But it ill preserve the patch naming section from MIDI SysEx and it will export that data. So external editors could edit this part of the patch. The default is that the 16 values are set to 0.

The feature is currently not documented, so we are effectively still totally free to specify the encoding, e.g. 
- 16 characters in 8-bit encoding (ASCII, UTF-8 variants, ...)
- 21 characters in 6-Bit encoding (64 different characters possible, more exotic)
- ...

Anyone writing an editor would need to know the encoding and once a decision is made it  needs to stay. Also: if it is ever intended to display the name, then the encoding must be very simple to implement in the firmware in order to save code size! Anyone up for some research and making an informed decision?

image-et-son, 2022-03-12:
Internally in the code, the numerical representation used for characters is UTF-8 in the range [32;126] (127 and anything below 32 is non-printable). So this will be the easiest to implement (e.g. no code overhead required). I would propose to make numerical 0 (which in UTF-8 corresponds to the NULL character ) to act as the end of the patch name. All other values to empty spaces.

We could produce a 32 character patch name from this (because the 8 bit cover the range twice).

What needs to be done now is the completion of a technical documentation, which covers topics like internal storage and SysEx spec. With that, someone with the skills to write a proper patch manager could implement the patch name, which - for the time being - would then only be accepted, stored and reproduced by the P600 but not yet displayed. 

I have already started a technical document ("...documentation/latex/p600fw_Technical_Reference.tex)") but it is work in progress.

matrix12x, 2022-03-12:
@image-et-son I was looking at several different encoding schemes and I agree that we should go with UTF-8 with zero representing the end of the patch name. 

I have never really dealt with Latex files before, how does one normally edit them?

image-et-son, 2022-03-12:
Latex (or Tex) is text file based and to get the desired output you "compile" it - it's a bit like mark-up language but much more powerful. The file based approach makes it particularly useful for repositories. Repositories don't like binaries because you cannot reliably recognized changes within the files. Latex also has the advantage that you can have a master document and you put in includes so you can have chapters in separate files. Also, it is very powerful. But then again, if you go beyond default setup it quickly becomes quite a lot to know, for example for creating tables etc.

Basically you edit the text files directly with the editor of your choice - but there are latex optimized editors which to syntax highlighing. The text file contains the plain text and commands which direct the layout. Commands always start with a back slash "\" and parameters (if any) are within curly brackets. For example you write "\chapter{In there beginning there light} text text text". And this will create a chapter heading and starts writing the text in a flowing paragraph. Many things can be set globally so you don't need to worry about it. You can define your custom commands (and I have made a ample use of that).

One way to contribute would be that you edit plain text and I do the layout and get it to compile.

If you want to get it to run you would need to install latex and a suitable editor. I have also made use of some special styles (available as optional install components). In particular I wanted a really fitting font which requires compilation by "lualatex" instead of the plain "latex" and the graphics are based on the package tikz. 

I can support you in getting this to run, but you need to make a decision whether you can spend the time. It will require some perseverance, I guess. 

Which part would you like to do?

Jpro600k, 2022-03-12:
https://github.com/mronetwothreefour/synth-midi-controller-plugins/tree/master/P-600-G
Someone has created a tool with a very nice interface. 
If he creates it, I think it will be the most correct path.

Jpro600k, 2022-03-12:
Is it possible to reproduce the randomize function on the hardware itself? 
in imogen OS?

closed 2022-06-17 by image-et-son