image-et-son/p600fw #142: noise waveform

Jpro600k, 2022-03-26:
Is it possible to change the current rough noise waveform to clean white noise?

Jpro600k, 2022-03-27:
Is pink noise more popular nowadays? 
Matrix12x is a noise board, and it seems that it used pink noise.
Is it possible to incorporate other noise waveforms while retaining the existing random noise? 
If you use white and pink noise without adding a noise board, it seems possible to create beautiful noise type tones. If that is possible, I think it would be attractive.

matrix12x, 2022-03-27:
Note, I did code pink noise as an LFO waveform a while back and as a LFO waveform, it "sounded" identical to the current "white noise" waveform. Pink noise is just low pass filtered white noise.

also, the noise source PCB I added is not a LFO waveform, it is another waveform for the VCO (so to speak). It is the random noise source from the TR909, which I think is white noise I may be wrong, though.

I put in a separate feature request to add the other LFO waveforms from the Matrix version of the code, which would be simple to do.

Jpro600k, 2022-04-26:
Will it be possible to keep both random noise and white noise?

Jpro600k, 2022-07-16:
If the white noise supported by the OS can be used without a noise board, the appeal of the OS will increase. It is good that the Prophet-600 does not need to be modified.

image-et-son, 2022-07-16:
Well, I have made some patches with noise from the LFO "noise" waveform, ie. some noisy pads and some East Enders or Simmons type tomtoms. But nothing can replace real noise. The best any firmware generated LFO can do is 2kHz. But maybe you can try to modulate osc B (only) using the noise wabeform and then mixing that in. You get somewhere but not quite there... Do you want me to share a patch for this?

Jpro600k, 2022-07-17:
I'll share my patch here first.
This is patch 91-99 I created in 2015 using the noise waveform of an LFO.
[LFO_Noise_Shape_Patch.zip](https://github.com/image-et-son/p600fw/files/9126805/LFO_Noise_Shape_Patch.zip)

Jpro600k, 2022-07-17:
I feel that the texture of the random noise waveform and the white noise are different. Clean white noise is essential for classic synth sounds.
 as I mentioned earlier, if the  clean white noise waveform can be used without hardware modification, the basics of sound creation will expand.
The more choices you have, the more fun your users will have.

You won't see the effect until you actually create a patch, but you'll probably get a breathtaking sound due to the different noises. Well, this is an imagination.
I don't know what the actual built-in sound will look like.

matrix12x, 2022-07-17:
@jpro600k the LFO noise waveform is white noise. It is literally the random function, which is what the code is doing. However, it is a LFO waveform, not a VCO waveform, so your perception of the sound may be different based on whatever you are modulating with the LFO noise waveform. For example if you use it to modulate the VCF or VCO, it will sound harsh.

I use the LFO noise waveform to make patches all the time.

I also use the noise source modification (parameter 44) to use noise in patches, like a VCO waveform, to make more traditional noise type patches, with clean white noise. For example, like snare patches, or noise sweeps using the filter envelope.

I kind of feel like you cross the idea of a VCO waveform with a LFO waveform in your description/discussion.

Jpro600k, 2022-07-17:
＠Matrix12x
So, is it a little off the topic of white noise?
There is no problem as it is. I believed in the possibility because I was saying that you could do it soon.

Jpro600k, 2022-07-17:
[os_lfo_white_noise.zip](https://github.com/image-et-son/p600fw/files/9127180/os_lfo_white_noise.zip)
[20220717_Prophet-600_rc1_2022.2_program_00_LFO_Noise_Shape.zip](https://github.com/image-et-son/p600fw/files/9127181/20220717_Prophet-600_rc1_2022.2_program_00_LFO_Noise_Shape.zip)

attached white noise preset file and sample
using LFO_Noise_Shape 

I don't use a noise board, but it's okay. The image of white noise is quite possible.

Jpro600k, 2022-07-18:
the LFO noise waveform is white noise.

closed 2022-07-18 by Jpro600k