image-et-son/p600fw #150: Alpha 15 - Stuck note on MIDI stop

RDPRDP, 2022-04-11:
I've been finding recently that when I press STOP on the DAW transport (Ableton) that the P600 will infinitely (no decay/release) hold a voice. I can only stop the voice by pressing the corresponding key on the P600 keyboard, then the decay/release envelope works.

GrantMTG, 2022-04-11:
Does the DAW send a note-off or other MIDI command to silence the synth? Do other synths (with a sustained patch) go silent? I'm not sure what is customary in this situation.

RDPRDP, 2022-04-12:
When pressing stop on the DAW three messages are sent (two Control, one Stop):

09:03:47.092	To Fireface UC Mac (23244188) Port 1	Control	1	All Notes Off	0
09:03:47.100	To Fireface UC Mac (23244188) Port 1	Stop		
09:03:47.103	To Fireface UC Mac (23244188) Port 1	Control	1	All Notes Off	0

I guess all notes off twice is just for good manners. At the moment I'm only seeing this problem on the current Alpha that I am on. I don't recall have this problem in the past. I also tried a different MIDI interface, but same result.

image-et-son, 2022-04-24:
Thanks, I will attend to it. Btw: "all notes off" is a feature that is new to the currently series. What is the behaviour in version 2.1 RC3?

RDPRDP, 2022-04-24:
I'll need to downgrade to 2.1 RC3, so this could take some time!

Jpro600k, 2022-04-26:
In beta1, there is no symptom that the sound keeps ringing when the DAW is stopped.
Have you tried using beta1?

RDPRDP, 2022-04-27:
So, after much testing the issue might be a change in relationship to my main MIDI interface (UM-880) on my M1 Mac Mini with Ableton 11. I'm trying different MIDI interfaces, with even different computers to try and see if I can make repeatable findings. More to follow.

RDPRDP, 2022-05-08:
After crazy amount of testing I feel confident that the below scenario from my MIDI interface will cause C4 to hold indefinitely (there is All Notes Off, but no off for C4). Pressing C4 on the P600 keyboard will release the voice back to normal. This is Mar 24 Alpha. I'll now try a few other versions as recommended above. At least now I know exactly what scenario to try and recreate.

13:58:24.118	To EDIROL UM-880 Port 1	Note On	1	C3	84
13:58:24.127	To EDIROL UM-880 Port 1	Note On	1	C4	72
13:58:24.272	To EDIROL UM-880 Port 1	Note Off	1	C4	91
13:58:25.000	To EDIROL UM-880 Port 1	Note On	1	D♯4	68
13:58:25.073	To EDIROL UM-880 Port 1	Note Off	1	D♯4	102
13:58:25.874	To EDIROL UM-880 Port 1	Note On	1	C4	78
13:58:25.909	To EDIROL UM-880 Port 1	Note Off	1	C3	64
13:58:25.909	To EDIROL UM-880 Port 1	Control	1	All Notes Off	0
13:58:25.917	To EDIROL UM-880 Port 1	Control	1	All Notes Off	0

RDPRDP, 2022-05-08:
Tested with Beta_1. Was able to recreate the situation. The C4 below became stuck. I unstuck it by pressing C4 on the P600 keyboard. I'll now try 2.1 RC3.

***NOTE: If you unplug a midi cable while a note is ON, should it sustain forever? That is also a scenario I experienced today, by mistake. Probably not an issue.

17:38:58.188	To EDIROL UM-880 Port 1	Note On	1	F3	84
17:38:58.197	To EDIROL UM-880 Port 1	Note On	1	C4	72
17:38:58.288	To EDIROL UM-880 Port 1	Note Off	1	F3	64
17:38:58.288	To EDIROL UM-880 Port 1	Control	1	All Notes Off	0
17:38:58.297	To EDIROL UM-880 Port 1	Control	1	All Notes Off	0

RDPRDP, 2022-05-09:
Just to clarify: I recognise Ableton is also an issue here, since it occasionally forgets to stop individual notes. I've opened up a ticket with their support.

image-et-son, 2022-06-17:
I don't really know what to do here. I cannot reproduce the issue.

RDPRDP, 2022-06-18:
Sorry for not updating this thread earlier. So, Ableton support confirmed I discovered a bug when playing two midi tracks at the same time with MIDI going to the same destination:

"I was able to reproduce the issue with a synth that doesn't handle CC 128.
If you stop playback after a note is played before the second clip plays the same, the note will hang.
 
Stop Playback on the highlighted area while playing 2 Clips with overlapping notes like these:"

https://imgur.com/V6MSf9z

https://imgur.com/5BFufGc

But, I think what this Ableton bug exposes is that the ALL NOTES OFF is potentially not work? Shouldn't the ALL NOTES OFF override the hanging note, and turn it off?

thanks!

RDPRDP, 2022-06-18:
Interesting: http://midi.teragonaudio.com/tech/midispec/ntnoff.htm#:~:text=Affects%3A,playing%20on%20the%20local%20keyboard.

I should try again with OMNI OFF it seems, just to be sure.

image-et-son, 2022-06-20:
Ah, I was looking in the wrong direction. I see that from the original GlIGli code, the assigner_allKeysOff() function ungates the each voice only if the key came from internal keyboard. Therefore it would just hold any external MIDI notes, which totally explains your observation. From my experience with the legacy code is that it is either something GliGli hadn't thought about (but that is more often associated with missing conditions rather than unwanted conditions) or there is a really good reason whay it was done that way. So I need  to think for second why this condition might have been necessary...

image-et-son, 2022-06-21:
Hi RDPRDP, would you mind trying out this version? It has the keys off MIDI command enabled also for MIDI notes. That should do the trick. Please remove and delete this version afterwards because it is "undefined" in the sense of the developement cycle.

[p600firmware_beta_2022-2_mod4RDPRDP.zip](https://github.com/image-et-son/p600fw/files/8952317/p600firmware_beta_2022-2_mod4RDPRDP.zip)
.

RDPRDP, 2022-06-22:
Sure thing, But my studio is torn apart right now, I will need a few days before I can get everything put back together again.

RDPRDP, 2022-06-26:
So far testing seems good. I'm unable to recreate the issue at this time. I'll keep trying a bit more, but I think it has been resoved.