image-et-son/p600fw #153: Hanging and silent voice

neodym1337, 2022-05-22:
Hi, loving the new functionalities of the OS so far! It's much more responsive and the envelopes feel more musical :) 

I do however have a problem with a hanging voice on mine. Using p600firmware_beta_2022_.1. When i start the synth one voice is sounding already. Droning with the same pitch unaffected by changes to paramenters. The other five voices are working like normally. By pressing 6 keys simultaneously i can sometimes catch the hanging voice to change its pitch, but it keeps sustaining after depressing the keys. After a few minutes as the hanging voice keeps going it drops in volume slowly until silent. After the hanged voice is silent its like playing with a five voice synth. After triggering the tuning procedure the hanging voice will keep on droning if it didnt go completely silent already.

It did not happen using gligli, was running 2.33 mostly

neodym1337, 2022-05-22:
It did some more troubleshooting and i did find that the stuck voice is sounding with both VCO A & B on Voice 6. It is actually affected by VCO Level parameter (for VCO A & B respectively) as well as wave form enable/disable. Also LFO to VCO modulation. So disabling all waveforms, or turning both VCO levels down mutes the stuck voice, and of course any other voices sounding through pressing keys on the keyboard. Unless smearing the keys its not affected by key note presses. It's also not affected by any filter or envelope changes. All other voices function normally

neodym1337, 2022-05-22:
I tried swapping the Filter and Vco chips between voice 5 and 6, but it made no difference. Voice 6 is still stuck. The stuck voice can also not be disabled by voice defeat.

matrix12x, 2022-05-22:
Any chance you know what the pitch control voltage on voice 6 is doing vs the other voices? Also, did this happen with any of the prior Alphas?

Are all the chips original, specifically I was wondering about the multiplexers.

@image-et-son could his issue be related to the glitches that happened to some synths?

neodym1337, 2022-05-22:
Hi @matrix12x thank you for your response :) I do know some multiplexers are replaced, the synth was refurbished (recapped, new PSU, several ICs replaced) a couple of years ago by a local highly rated professional and has been working well since with both original and gligli firmwares.

I think 082 x3, 4051 x3, 4016 x1, 4053 x1 on voiceboard 1 had to be replaced.

Didn't try the alphas, will do it now :)

No didnt check the pitch control voltage yet, but i have a DMM and could try. I took a look at the CEM 3340 pinout, would it mean checking pin 15? If i got it right that's freq CV in.

Cheers

matrix12x, 2022-05-22:
Can you let me know all the markings on the 4051 chips.

neodym1337, 2022-05-22:
For sure, 
U415 - CD4051BE B7W082FE3
U416 - HD14051BP OM4 4
U417 - HD14051BP OM4 4
U418 - BU4051B 401 - 221

U418 is original.

I also found that alpha 12.2 (p600firmware_v2.x-alpha-12.20220119.syx) does not have the issue. But the next version 13.6 does.

image-et-son, 2022-05-22:
> I tried swapping the Filter and Vco chips between voice 5 and 6, but it made no difference. Voice 6 is still stuck. The stuck voice can also not be disabled by voice defeat.

> I also found that alpha 12.2 (p600firmware_v2.x-alpha-12.20220119.syx) does not have the issue. But the next version 13.6 does.

Hi, neodym1337, thanks for posting and thanks for also checking the alpha versions. Such information is super helpful when trying to narrow down the cause. At first sight it does seems like it could be related to the multiplexer. Is the following correct?
- Reacts to VCO frequency change by LFO but not by key press (and presumably also not by poly-mod to osc A freq?)
- Reacts to VCA change by volume knobs (and LFO/VIB?) but no via envelope
- All this applies to both oscillator A and B

If that is true then it hints that the high frequency (e.g. 2kHz) voltage setting is too fast on that output channel because all the voltages which work are set in a "slower" function. What surprises me is that it would affect one voice entirely rather than only a particular function. It also hints that it is a hardware "problem" - having said this, it depends on hard hard you drive the hardware and since older versions work, it is a question whether the new software drives it too hard and how many people would be affected by this. I would suggest that I send you a trial only version with a slower update to see if that works. Maybe you also have the possibility to change the chips?

matrix12x, 2022-05-22:
@image-et-son I agree with your analysis.

neodym1337, 2022-05-23:
@image-et-son thank you for your response!

* Reacts to VCO frequency change by LFO but not by key press (and presumably also not by poly-mod to osc A freq?) - Yes does react to VCO frequency change by LFO, but not key press. Does react to OSC B to OSC A freq polymod, but not filt env.
* Reacts to VCA change by volume knobs (and LFO/VIB?) but no via envelope - correct
* All this applies to both oscillator A and B - correct

Really interesting that the setting might be too fast! Just curious of you got any clue 

Yes, i definitely want to try to change the chips. You are referring only to the 4051? Before your post found in the old gearspace thread an experiment from @matrix12x to replace the original 4051 with these ones https://www.mouser.de/ProductDetail/595-CD74HCT4051E Since the three of them are already replaced and socketed i only have to desolder one, so i think i will try to order them. 

Yes i would love to try a trial version.

image-et-son, 2022-05-23:
Hi there, please try the version in the file. At this stage I have not changed any "timing" things but I have changed the order of voices in the update cycle. If you try it, does the problem change from voice 6 to voice 5? It may or may not change but the result will help me to pinpoint the root cause.

[p600firmware_beta2022_mod4neodym1337.zip](https://github.com/image-et-son/p600fw/files/8752999/p600firmware_beta2022_mod4neodym1337.zip)

BTW: this a pure technical test version. The best is probably to reinstall an "official" version after the test and then delete version files so that there is no risk of confusion.

neodym1337, 2022-05-23:
@image-et-son Just tried it, the problem did change from voice 6 to voice 5

image-et-son, 2022-05-23:
> @image-et-son Just tried it, the problem did change from voice 6 to voice 5

Ah! Thanks, that's interesting. It probably still means that a faster chip could solve it (like it works on most P600s) but it gives me a hint where to look for next in the software. 

And I can already think of a brute force solution that would remedy similar hardware situations: cycling through the update order. This would certainly do the job for you. But maybe there is another solution to this. Let's see...

image-et-son, 2022-05-24:
> For sure, U415 - CD4051BE B7W082FE3 U416 - HD14051BP OM4 4 U417 - HD14051BP OM4 4 U418 - BU4051B 401 - 221
> 
> U418 is original.

Hi, I am trying to put together small summary of what we found when comparing different 4051 variants, although much of it is heuristics. Still, the data sheets can be used to get an idea of how fast the chips are. From that I can tell that the HD14051BP is a slow chip, almost in the range of the MC14051B which we found to be prone to glitches. I saw MC14051B refitted on another P600 but I don't know if there's pattern to it. It is certainly not a good choice, as is the HD14051BP.

The CD4051BE (which I have originally in my P600 for 416-418) is OK. Much better is MC74HC4051N (which Matrix12x pointed out). From datasheet the BU4051B is probably in the range of the CD4051BE. You have this for U418 - btw: this one does not control any of the voltages which are "missing", only slow ones, so I would not focus on it. Btw: I have a fairly slow HCF4051BE for U415 but it seems to get the job done.

image-et-son, 2022-05-25:
OK thanks. I have another one. This one is "broken" on purpose. It will not respond to pot changes at all. It will pick up the master volume upon start up but then not responds. You can still use the buttons, for example to select a patch. The questions is: do all voices play normally in this version? 
[p600firmware_beta2022_mod4neodym1337_3.zip](https://github.com/image-et-son/p600fw/files/8769619/p600firmware_beta2022_mod4neodym1337_3.zip)

neodym1337, 2022-05-25:
Thanks for continuing to troubleshoot! I tested [p600firmware_beta2022_mod4neodym1337_3.zip](https://github.com/image-et-son/p600fw/files/8769619/p600firmware_beta2022_mod4neodym1337_3.zip) and it seems it behaves like the previous v2 version, except of course for the mentioned non responsive buttons. Voice 6 is still stuck, but is responding to note change.

image-et-son, 2022-05-25:
OK, and thank you for going along with the testing. I always learn something new too. I hope I can get to the bottom of it. Obviously, there is  no interference with control read outs. I will come with another version to test but I am off for the next 4 days so I don't know if it'll be ready before the end of the day or not. BTW: Do you notice any changes after the instrument has had time to warm up? Is it getting better, if only incrementally?

neodym1337, 2022-05-26:
Sounds good, no sweat! :) Let it warm up for +1h but didnt notice any difference. Will experiment some more, got the multiplexers coming next week as well Cheers

neodym1337, 2022-05-31:
With new multiplexers everything is working! Huge thanks for the support

image-et-son, 2022-05-31:
> With new multiplexers everything is working! Huge thanks for the support

Which ones did you put in? The "MC74HC4051N"?

matrix12x, 2022-05-31:
@neodym1337 awesome!!

image-et-son, 2022-06-01:
> I ended up using the CD74HCT4051 (Texas Instruments brand) in the end. I have them in sockets. They work great. I get zero weird effects and very low latency from keypress to note on.
> 
> I was going to use the MC74HC4051 series. I forget why I didn't in the end. I think it was parts availability at the time.
> 
> Also, I don't know if you wanted to try and make a version of the code that really pushes the multiplexers to their breaking point (e.g., maybe 3 kHz refresh or something). I'm game for trying that out. Although, It seems like when people swap out parts they seem to do it willy nilly and often put in "bad" (I.e., slow) 4051 chips.

Hi, I tried a higher frequency. But you remember the tests we did with el-folie? When I reduced the wait cycles I could even drive the ok-ish 4051s to glitch. Also yours did that, if I remember correctly. So I fear, that's the end of the line. For 3kHz it would need to be considerably faster.

image-et-son, 2022-06-01:
Not a real bug. glad everybody's P600 is working with the new firmware :-)

closed 2022-06-01 by image-et-son