image-et-son/p600fw #157: Separating spread for detune and envelope variance

neodym1337, 2022-06-01:
Loving the variable spread parameter. With one caveat, which is a very personal preference, but i find the osc detuning when using spread a bit too strong in relation to envelope variance. On one hand i like the simplistic choice of having one parameter to control both at the same time, but it feels a bit limited as well. How about having them separate? Cheers

image-et-son, 2022-06-02:
Hi, we did some iterations of testing. Do you think we could just try to reduce the strength of the pitch variation slightly? I could do that in the new beta for the release. A new and split parameter would be too drastic a change at this stage (and we're running out of MIDI CC slots, too). 

BTW: I think pitch shift could indeed be too strong relative to envelope variation for small vintage values and short envelopes but at larger values and/or longer envelope times it is the other way round, I think.

neodym1337, 2022-06-02:
Sounds like a good compromise! And i can agree on it being a bit the other way around at higher settings with long envelope times.

image-et-son, 2022-06-02:
> Sounds like a good compromise! And i can agree on it being a bit the other way around at higher settings with long envelope times.

OK, I'll send you a specific version for testing the amount. I don't want to publish a second beta or new alphas, so you'll receive a private version for your request and when it's ok it'll go in to the release.

image-et-son, 2022-06-02:
What's your overall feeling with "vintage spread". Is it a strong effect on the control travel? Where would you put it typically? Reason for asking: if I turn down the spread on pitch further I will get only a resolution of 16 values from 0 to 99 so I'm tempted to increase the strength of the envelope variation instead. But this increases the overall effect. What do you think?

neodym1337, 2022-06-06:
What i usually do is put the oscillators as in tune as i can, and then use the vintage spread to get an acceptable detune for the type of patch i'm programming. I do it mainly to get the envelope spread, but since the detune has a lot of impact on the sound i let the detune be the limit. Often i end up around 40-50

matrix12x, 2022-06-10:
@image-et-son I agree with this statement
"BTW: I think pitch shift could indeed be too strong relative to envelope variation for small vintage values and short envelopes but at larger values and/or longer envelope times it is the other way round, I think."

closed 2022-06-17 by image-et-son