image-et-son/p600fw #169: beta2 suddenly jump to the program with the current value of the knob.

Jpro600k, 2022-06-18:
For example, if you start the sequence with program99 and turn the knob while holding down the from tape button, the program number will suddenly jump to the program with the current value of the knob.

After reading the current program number that started the sequence, the program number cannot be switched smoothly from that value.

Starting the sequence, the knob is reading the speed, but it is a problem that the number at the dial position of the speed is applied to the program number.

If you turn the knob while holding down the from tape button without using a sequence, you can move the program smoothly from the current program number.

image-et-son, 2022-06-18:
I wonder if the "select patch by data dial" is more trouble than good. I'll see what I can do.

Jpro600k, 2022-06-19:
I don't think it's correct to have different behavior.

Knobs are used to select pre-arranged programs as you progress when playing with a sequence.

This is useful for intuitively searching for the one that fits the track while playing a sequence of peripheral programs.
The sudden jumping of the program disturbs the mood of the performance.

image-et-son, 2022-06-19:
Hi, would you strongly mind if I removed the patch selection via dial from the current release and bring it back properly in a subsequent release? I am unhappy with the way it is but it would require too mich testing to reprogram it. So I would like to postpone it.

Jpro600k, 2022-06-19:
I think it is better to maintain the program change with the dial function as it is.
The procedure in question is clear. It may be a special playing procedure.
I don't think it's a problem with the release.

Jpro600k, 2022-06-19:
There is no problem closing this ticket.

image-et-son, 2022-06-19:
OK, we'll revisit the thing after the release.

closed 2022-06-19 by image-et-son