image-et-son/p600fw #170: beta2 the preset LED will go out and suddenly go into live mode.

Jpro600k, 2022-06-18:
When a program change is received from an external synth, the preset LED will go out and suddenly go into live mode.

image-et-son, 2022-06-19:
Hi, I just tried it. When a program change is sent to the Prophet 600 then this is accepted only in preset mode and I found that the preset mode is not changed to live mode. So there must something more to your ticket. The Prophet 600 does also react to mode change by MIDI CC. This is MIDI CC = 0. Maybe you also sent a mode change event?

But thank you for the ticket ! I discovered a bug in this way: when you send a prog change for a program that doesn't exist the current program remains on display but the default patch is loaded.1 I just fixed that and that will be fixed in the release. Thanks!

Jpro600k, 2022-06-19:
I was able to confirm that MIDI CC = 0 was sent.
It's okay to close this ticket.
I think it's better to open the bug you find yourself with another ticket.
Thank you.

image-et-son, 2022-06-19:
OK, thanks.

closed 2022-06-19 by image-et-son