image-et-son/p600fw #177: More than half of the knobs seem to have no change in noise volume. Is this normal?

Jpro600k, 2022-07-11:
I modified the noise. I succeeded in mixing noise at to tape 44.
I was wondering if I could hear more intense roaring noise, but it wasn't.
More than half of the knobs seem to have no change in noise volume. Is this normal?

image-et-son, 2022-07-12:
I think that Matrxi12x should comment on this. Maybe I need to change the travel. But I'd need more feedbakc on this.

Jpro600k, 2022-07-12:
it does mean the response of menu option 44. speed knob.
speed knob value 50-99 sounds almost the same. 
From 50 to 99, can you hear the difference clearly?
I think the voltages are probably not the same. I don't know because I haven't measured it. Your opinion will be helpful. Thank you.
I may also solder to input B.

matrix12x, 2022-07-12:
If you want I can measure the control voltages at different settings.  Let me know.  Maybe we should make the response curve log

Jpro600k, 2022-07-13:
I would like you to change it so that it changes smoothly up to the value 99.

Jpro600k, 2022-07-13:
Thank you for your wonderful survey. After a while, connect to input B as well.

> So I did some measurements. Pin 10 of U458 at parameter 44 value of 50 is about 2.5V and at 99 it is about 5V. The nose when measured at the output of the synth plateaus at about 1.75V p-p at parameter 44 value 38-39 or so (which seems to match GliGli's values above).
> 
> Accordingly, 24576 should be the max value. I suggest making this code change.
> 
> Also of note, @Jpro600k if you need more volume maybe hook up 100K resistors/noise to the B side also. It would increase your noise volume by 3dB (double).

Thank you for your wonderful survey. After a while, connect to input B as well.

image-et-son, 2022-07-14:
Hi, here is a trial version with the external voltage control (i.e.noise level) scaled to the range we have identified. Please check it out and tell me if it is ok. And then don't copy, reuse or distribute this firmware version - it is not a tracked version.

BTW: it has all the fixes after beta 2 in it and also the one new feature: LFO Trigger (parameter 222) "Key". This retriggers the LFO with a new key stroke when no other key is held (similar to modulation delay). I think this is very useful :-)
[p600firmware_beta_2022_2_fix177.zip](https://github.com/image-et-son/p600fw/files/9114231/p600firmware_beta_2022_2_fix177.zip)

matrix12x, 2022-07-14:
I confirmed LFO trigger works. but it is located at parameter 1 1 1 , and I thought it said 'Hey', not 'key'. I was excited. lol
also, the ext voltage control has no additional effect after about 50. it works fine from about 0 to about 50. I watched the noise level on a scope on the output while turning the knob (parameter 44).

image-et-son, 2022-07-14:
So the noise travel doesn't work!? Will look into it tomorrow.

matrix12x, 2022-07-14:
correct. If you want I can measure the control voltage at pin 10 at mid and max range. but I think at max range we want it to be 2.5VDC, based on my testing not the full scale 5V.

image-et-son, 2022-07-15:
Haha ... the change was such a no-brainer that it made me sloppy. For the change I copied the old line of code where the voltage is set and then forgot to delete the old line. And then since it came after the new one....

Here is the corrected one.
[p600firmware_beta_2022_2_fix177_2.zip](https://github.com/image-et-son/p600fw/files/9118753/p600firmware_beta_2022_2_fix177_2.zip)

Jpro600k, 2022-07-15:
> Also of note, @Jpro600k if you need more volume maybe hook up 100K resistors/noise to the B side also. It would increase your noise volume by 3dB (double).

Hi, Matrix12x

I also connected to B. 
The amount of noise is definitely increasing. Very close to ideal.

matrix12x, 2022-07-15:
I have this steps too, but this seems to be more of a set and forget setting than a live performance tweak setting, so I didn't care.

Jpro600k, 2022-07-15:
After updating to fix117_2 , I noticed that the tuning was significantly out of sync. Turn off the power, wait a while, short the TP301, and try tuning again.

Jpro600k, 2022-07-15:
> I have this steps too, but this seems to be more of a set and forget setting than a live performance tweak setting, so I didn't care.

Is there a possibility that it can be done smoothly?

Jpro600k, 2022-07-16:
I removed the noise modification, restored it to its original state, and then tuned it, but the pitch is random. Is there a way to fix it?
sound test
[tuning_issue.zip](https://github.com/image-et-son/p600fw/files/9125155/tuning_issue.zip)

> Haha ... the change was such a no-brainer that it made me sloppy. For the change I copied the old line of code where the voltage is set and then forgot to delete the old line. And then since it came [...]

About beta_2022_2_fix177_2

I removed the noise modification, restored it to its original state, and then tuned it, but the pitch is random. Is there a way to fix it?

Check this.
[tuning_issue.zip](https://github.com/image-et-son/p600fw/files/9125159/tuning_issue.zip)

Jpro600k, 2022-07-16:
> After updating to fix117_2, I noticed that the tuning was significantly out of sync. I turn off the power, wait for a while, short the TP301, and then tune again.

I tried to adjust the pitch by the TP301, but the display is random and unstable, it works only in the range of 99 to 40, and I can not approach 0.0.
The noise mod is reattached.
The LED is displayed normally when booting in normal mode. There is no random display.

matrix12x, 2022-07-16:
@jpro600k I think the synth does not "mute" the noise mod properly when tuning. 

NOTE: My noise mod has a volume trimer one the output. I found I have to manually turn down the noise mod to almost zero volume then tuning works. 

Its weird because the code looks like it should mute the noise during the tuning sequence. This should probably be spun off as another bug. But it's a long existing bug, I know my code had the issue.

image-et-son, 2022-07-16:
Yes, I looked it up as well: the noise level (as corresponds to the menu parameter 44) is set to zero in preparation of tuning. I have not experience with the noise mod, though. So I don't know if it somehow still enters the tuning procedure even if set to zero.

matrix12x, 2022-07-16:
I may at some point monitor the voltage on pin 10 and see what happens during tuning.

closed 2022-07-16 by image-et-son