image-et-son/p600fw #182: When I send snapshot from Ctrlr panel, The LED has turned off.

Jpro600k, 2022-07-24:
When I send snapshot from Ctrlr panel, The LED has turned off.
example CC39 and CC40
is this bug?

Jpro600k, 2022-07-26:
Is it the result of the inconsistency between the editor tools and the latest OS?

image-et-son, 2022-07-27:
Hi, you need to fill me in on some things. I don't understand yet. What do you mean by "snapshot"? And what is "Ctrlr panel"? 

MIDI CC39 and CC40 are amplitude and filter velocity of the patch. Is that what you intend to change?

Jpro600k, 2022-07-27:
Hi,
The tool is here.
Prophet 600 CTRLR panel and AU (Marix12x's  tool)
https://github.com/matrix12x/p600fw/blob/master/Prophet%20600%20CTRLR%20panel%20and%20AU.zip

MIDI CC39 and CC40 are example.

matrix12x, 2022-07-27:
the panel was based on the CCs for the last version of my OS. so I'm positive there will be some issues using it on Imogen's OS. Also, I'm not updating the panel yet.

image-et-son, 2022-07-28:
OK, obviously, there are more controls now. However, I conciously did not reuse redundant CC slots for new functions so that at editors should at least be compatible (even though some functions, such as the LFO range would have no effect). This means new CC functions are in a range that previously not used. I wonder what compatibility issues you may encounter.

Jpro600k, 2022-08-06:
I would like to update the panel, but the compatibility with the latest OS was not expected in the first place, so the ticket will be closed.

closed 2022-08-06 by Jpro600k