image-et-son/p600fw #184: Bug - tuning error for units with noise mod.

matrix12x, 2022-07-27:
On all versions of the OS (including prior gligli ones as well as my own), for any P600 that has the noise mod (a separate noise source fed to Input A/B of the 3372) the P600 looks like it tunes, but the results are a unit that is out of tune. I have a feeling this happens to the VCF tuning in unmodified units where a ctrl voltage is fed into the external CV input (needs to be confirmed). 

The temporary fix is to have an output level trimmer on your noise source PCB and when you need to tune, temporarily turn the noise level to zero. This seems to work great for tuning, but is super inconvenient.

I monitored the control voltage to pin 10 on U458 during tuning and the control voltage appeared to stay at zero (which is correct). The tune algorithm mutes this control voltage. I did not monitor the output of the U458 for noise level. I should.

Next, I fed the noise only into input A (not input B) on the 3372 and hit tune. Result, VCO A and the VCF (on res 99, keytrack on full) (as a sine wave), were wildly out of tune, but VCO B was in tune.

I will probably monitor the VCO A waveform and see if there is some noise bleedthru that is mildly effecting the zero crossings of the VCO A saw during tuning. During tuning the saw drives a flip flop. I'm wondering if somehow the noise circuit effects this. 
I will probably also monitor the flip flop. I'm wondering if some additional wait cycles or delay to the tuning algorithm would help.

image-et-son, 2022-07-27:
Are you confirming that from the software side, the control voltage is correctly set to zero during tuning? Please check the noise level also. Do you have a circuit layout for the noise mode which you can share? Ie. I want to know where the noise is fed into the signal path. Thanks

matrix12x, 2022-07-27:
I confirm that on the software side, based on my measurement of the control voltage, that the control voltage seems to be set to zero (or almost zero) during tuning. I will take a precise measurement, I was eyeballing my scope during tuning, so if it was a few mV I would not have noticed. As noted above I need to measure actual noise output from U458. My current theory is there is still some amount of noise (small) that is leaking at min noise CV (extfiltCV).

The noise is fed into the signal path at pin 6  (input A) of each of the 3372 via a new 100K ohm summing resistor.
I don't have a layout or schematic of the noise circuit or the mod. The noise circuit I used is the same schematic as the noise source of the TR909 though, with a 10K trimmer on the output.

As for the mod, the input resistor to U458 is lifted, disconnected the external CV jack (which is then jumped to the equivalent of, for example, R417 for each voice, filter CV in for each 3372). Then the noise source is patched to the input of U458. The output of U467 (pin 1) is disconnected from, for example R417 (or equivalent) for each 3372, and gets connected to a new 100K ohm series resistor that is connected to pin 6 of each 3372.

matrix12x, 2022-07-27:
I should add to this info that both me and Jpro600K have the Panorama mod, which does rewire the outputs of the 3372 chips (pin 14) and has them go to the panorama mod board. So, that may add to the complexity of the issue. However, I'm not sure yet.

So I measured the output of the VCA that controls the noise level (CEM 3360) U458, and when set to zero volts CV (parameter 4 4 value 00) there is still noise present at the output. (approx 5mV or so.) I'm positive that this is the issue. I think the bleed thru noise is confusing the flip flop when it is counting.

Schematic for the noise source mod.
[image]

matrix12x, 2022-07-27:
Also, of note, the data sheet for the CEM 3360 shows a 0 to 2VDC control voltage range. which seems to make sense in light of the recent request to reduce the range of parameter 4 4 (ExtFilCV).

matrix12x, 2022-07-27:
Yep, it is the bleedthru. 
I experimented by setting the output level of the noise source at resistor S4219 in the above schematic to various level and hitting tune. (also looking at the noise level at pin 1 on U467 for reference)

I reduced the noise level using the trimmer on the output of the noise source so that the voltage p-p at resistor S4219 in the image above I measured max 1.4V p-p of noise (I was running at about 5V p-p at this point previously. This was the largest level signal that I could get a good tune with. thus, this bug should be closed, but we may want to make a note in either the code or the manual. I'll post on gearspace so anyone else that added this mod and has tuning issues has a solution.

image-et-son, 2022-07-28:
Do I correctly understand what your're saying the following: "You need to trim the output on the noise source because if you exceed a threshold the minimum output interferes with tuning."

matrix12x, 2022-07-28:
yes

Jpro600k, 2022-07-28:
Attach a 50k trimmer to the output of the noise board, set the volume to 0, and then tune with the TP301. I was able to tune normally and there was no problem.
From now on, I will always adjust this way.
As mentioned above, there is no need to change the OS.