image-et-son/p600fw #185: Add "Mix Velocity" Parameter

image-et-son, 2022-07-31:
This would provide even more expressivity as it changes the timbre

matrix12x, 2022-07-31:
Great idea. Would a velocity of 64 be middle mix?
Or would this be a "bipolar" parameter where velocity below 64 subtracts and above adds?

image-et-son, 2022-07-31:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;} When I thought about it imagined the latter.                                                                                                                                             Sent via BlackBerry Hub+ Inbox for Android                                                                                                                                                                           From: ***@***.***: 31 July 2022 17:24To: ***@***.*** to: ***@***.***: ***@***.***; ***@***.***: Re: [image-et-son/p600fw] Add "Mix Velocity" Parameter (Issue #185)  
Great idea. Would a velocity of 64 be middle mix?
Or would this be a "bipolar" parameter where velocity below 64 subtracts and above adds?

image-et-son, 2022-08-11:
Add info. Same "thought trap" again: since "mix" is a parameter that affects all coices, it cannot be implemented in that way, unfortunately.

closed 2022-08-11 by image-et-son