image-et-son/p600fw #187: Functionality to calibrate poly-mod oscillator A

image-et-son, 2022-08-02:
I find that on different units the strength of the poly-mod FM (i.e. target oscillator A) varies from voice to voice. This makes harmonic sounds with strong poly-mod FM very difficult. Since there is no hardware adjustment for this, the idea is to introduce a calibration function (calibrate by ear) with data stored in the settings. This should be accompanied by a calibration standard, ie. defined frequency and ploy-mod amount and then adjustment such that the result is harmonic. This could be done voice by voice. It would be major addition, though.

matrix12x, 2022-08-03:
The polymod FM circuit in the P600 has a hardware design flaw. It should have been built with a series cap as shown in the CEM 3340 datasheet. The DC offset screws up the frequency you get when doing FM. the different VCO wavefoirms each have a different DC offset and each VCO has a different DC offset.

I added caps in mine like in the datasheet.

[image]
[image]

image-et-son, 2022-08-03:
OK, I see. Does this solve the issue? I mean, do all of your voices behave the same with this modification?

I wonder, if you have a modification like this, doesn't your unit sound different? I would expect that patches with moderate to high oscillator FM (and no sync) sound different on you unit, potentially horribly out of tune. Is that true for the two attached examples? Or can you "repair" those with a little change in fine B tuning or poly-mod osc B amount?

[poly-mod_patches.zip](https://github.com/image-et-son/p600fw/files/9248656/poly-mod_patches.zip)
.

matrix12x, 2022-08-03:
I’ll try it out soon.  I think it’s a poor design choice of Sequential to use the vco cv input for fm instead of the fm input for the chip. And to leave out the dc block.  By doing this any dc offset is a VCO cv.

matrix12x, 2022-08-03:
Tested. Yeah, those are both really out of tune.
on patch 71, I had to change OSC A frequency to 18 from 35 (or whatever) to get it in tune.
Actually same with patch 90.

image-et-son, 2022-08-03:
Thanks - that's quite a setback for the idea of the patch collection and it needs to be documented as well. I would not say that patches should necessarily accomodate instruments with your mod but I recently bought a second unit and while on my first one the voices are quite (but not perfectly) even, on the second it is all over the place. I think that having strong FM patches is just not a good idea for a collection because it is not reproducible and it sounds not just different but potentially ugly.

For the calibration I thought about adjusting (scaling) the poly-mod amount. I would generally accept the drift in VCO A CV with poly-mod, but then 1) define a fixed point (ie. set  A and B to prescribed values and then change poly-mod until its harmonic) such that the result becomes comparable between units and 2) have all voices adjust individually so that is sounds clean. In your modified case it wouldn't be possible or meaningful anymore. It is a different way of dealing with the problem but it seems to alter the instrument significantly - the difference between 35 and 18 is huge.

PS: I will have my second unit modified with the capacitors. Might come back to you for details.

matrix12x, 2022-08-04:
I agree re: not worrying about modified units.
I tested this on my unit immediately after turning on. I can let it warm up for 30 min and see if the tuning difference is the same. I'm sitting here thinking about it and I'm wondering if I overshot and went to a different harmonic than you did, since I did not have an audio reference of your good version. Maybe post here a wav or mp3 for me to compare to for patch 71.

The mod is easy:
Lift one leg of R430 and add a 0.1 uf cap (in te end I used leaded caps instead of tombstoned surface mount caps) between that leg and the hole you lifted it out of (and R480, R429 etc. Look at the schematic in the service manual for the other ref designators).

Here is my blog post on the mod, including spectrum plots showing the effect of the DC offset on the freq of OSC A:
https://prophet600noisemod.blogspot.com/2020/10/prophet-600-poly-mod-vco-b-to-modulate.html

matrix12x, 2022-08-04:
ok, patch 71, warmed up for 30 min, changed VCO A from 35 to 36 and VCO B fine tune to -5 and the beating and out of tune went away, mostly, my tuner showed VCOA to be a Bb when I pressed any C. 
At a VCOA of 48 and a VCOB fine tune of 00 I am at C when pressing C.

I think I was tuning to a diff harmonic at 18, but a VCOA of 15 gives me a C when I press any C.
for patch 90, VCOA of 18 is the only thing that gave me a C at pressing C

image-et-son, 2022-08-04:
> ok, patch 71, warmed up for 30 min, changed VCO A from 35 to 36 and VCO B fine tune to -5 and the beating and out of tune went away, mostly, my tuner showed VCOA to be a Bb when I pressed any C. At a VCOA of 48 and a VCOB fine tune of 00 I am at C when pressing C.

You mean 38, right? (not 48)

I never really paid a lot of attention to the topic, but I now see it clearly before me. Thanks for explaining it. I did notice the following: 
- DC offset difference among the voices mostly comes from saw and pulse, much less from triangle. Obviously you have the overall drift when you change the pulse width. Especially from that perspective I totally agree that it is a very bad choice to feed that signal into the CV in. The pulse wdith produces an offset.
- I managed to remove most of the voice variation on my second unit by scale adjusting it carefully. The units sound pretty much the same now. Therefore, you cannot rely only on tuning. The scaling adjustment has an immediate impact on the FM behaviour. For me, this saves the idea of the patch collection. I will need to document it. (I will eventually mod my second unit anyway.)

I do think that FM calibration (either through adjusting poly-mod amount or adjusting VCO A) might still be a super cool feature because it could lead to a new sound quality.

matrix12x, 2022-08-04:
You are correct regarding the pulse and saw. 

Interesting enough, the linear FM of the CEM chip input is linear, the CV input is exponential, so I wonder what effect this has on the sound. I may modify one voice to try out the linear FM input.

Re: my settings, it should have been 38, that would make sense, but for the tuner to show a C for VCO A it was 48 (not 38). I did not look at the fundamental on a spectrum analyzer, so I wonder if the tuner is zeroing in on a harmonic.

I will investigate further.

matrix12x, 2022-08-05:
ok, watched it on a scope, didn't realize I could hit notes "between" numbers, for example, if I went to between 36 and 37 (VCO A) I could get C to be C. I watched it on a spectrum analyzer.

Poly mod OSC B amount does shift the freq (vs without my cap mod), but not as much as before the mod. So this P600 of mine may still benefit some from your proposed FM calibration, which seems like an awesome idea.

I have a CEM 3340 VCO for my eurorack, and it has the FM and CV inputs as patchable points, so I may try that VCO out instead of modding my P600 to see if the linear FM input causes less of a shift. The cap mod seems to have made all of the waveforms act the same in terms of any shift. Saw, pulse, and tri all work the same in terms of shift. they do make different overtones though,a s expected.

image-et-son, 2022-08-05:
I wonder if I can even utilize the internal comparator which are also used for tuning!? 

I guess the complexity comes from the interdependence of the parameters, ie. the direct correction of VCO A would depend on poly-mod amount, on VCO B pulse width and potentially also on VCO B frequency. With correcting the poly-mod amount there is some hope that it is less dependent on VCO B pulse width. Then again, might be wishful thinking.

image-et-son, 2022-08-08:
> I wonder if I can even utilize the internal comparator which are also used for tuning!?
>
> I guess the complexity comes from the interdependence of the parameters, ie. the direct correction of VCO A [...]

Correction to this. I frequently forget that some parameters are not available per voice. This includes poly-mod amount and PW. So to make per voice corrections it will have to be oscillator frequency A.

image-et-son, 2022-08-14:
I use this thread for documentation. I did some data analysis but changing parameters and then measuring the movement of oscillator a pitch. In this way I hope to devise a simple corrective function that compensates for non-calibrated differences in the voices. The idea is not to remove the dependency of oscillator A frequency in poly-mod but to normalize it, i.e. set a should be relation for afixed set of values and then measure the voice specific deviations around it so that it can be corrected. The correction function mustbe VERY simple. It seems that this can be done. Here are my "experiments" and my preliminary results:

**I changed the parameters:** 
- oscillator B shapes: SAW (ON/OFF), TRI (ON/OFF), PULSE (IN/OFF) + PW Range 10 to 90
- Poly-Mod OSC B amount (with target freq A on, fo course)
- Target Octave difference A/B: 0, +1, +2
- Base note of several octaves  

**Results to far**
- The pitch shift of OSC A is independent from base note (phew! Anything else would have been a show stopper)
- The absolute pitch shift of OSC A as measured in the "frequecy A" parameter  is linear in poly-mod amount. However, there is an offset in poly-mod amount of about 9.300 below which there is no effect. If that threshols is voice dependent hasn't been analysed yet.
- The linear relation of pitch shift of OSC A and poly-mod amount is independent from the octave difference (interesting! I was expecting that I would need to apply a relative shift or something. But the result is absolutely robust)
- The different oscillator shape contributions change the linear relation of pitch shift of OSC A and poly-mod amount. This was already known, in particular that the PWM changes the pitch.
- While the relation of PWM to OSC A pitch movement is also linear (again good news!) the effects of the different shape contributions (shape switches) are additive, i.e. the offset created by SAW + TRI is less than the offset of SAW plus the offset of TRI, etc. But I think that can be handled because there only a few combinations.

In the end we will have a corrective function for frequency A of the form: Delta(FreqA)= (PolyModAmt - 9300) * (ShapeOffset + PWOffset * PWB)

The two parameters to be determined in a calibration procedure are ShapeOffset and PWOffset. ShapeOffset could be a set of numbers relating to different combinations of OSC B shapes. Ideally the calibration would be done using the hardware tuning infrastructure.

So all in all it looks doable in ony way or another. If it will be practical that is another question. Also: the curent firmware would need to be optimized because there is not enough space on the Teensy++ at the moment).

image-et-son, 2022-08-14:
Next set of results:
- As stated before there is a linear relation of PWB to OSC pitch A. I now find that for a given poly-mod amount this relation (i.e. the slope) does not depend significantly on whether TRI and/or SAW is active at the same time. In othter words, the shape switches on OSC B produce constant offsets in OSC A pitch which then only depend on poly-mod amount but not on PWB.
- However, as stated before the offsets of the wave shapes are not additive. Therefore, together with the PWB/pitchA relation (2 measurements at a given poly-mod strength and only pulse active) 6 more measurements are needed: 1) offset only TRI, 2) offset only SAW, 3) offset TRI and SAW, 4) offset TRI and PULSE, 5) offset SAW and PULSE, 6) offset SAW and TRI and PULSE. For 4, 5 and 6 it is sufficient to do this for one value of PWB.       
- So the solution will be an 8 parameter per voice calibration which requires interpolation only when PWB or poly-mod amount is changed (performance cost of that calculation is to be determined - could be tricky for LFO PWM, but that could also be left out in the first attempts).

matrix12x, 2022-08-14:
Maybe remove per note tuning to fit this code.  Seems like much better functionality if you can get it to work.  

Also do you want me to test my synth for these same measurements

matrix12x, 2022-08-15:
I think polymod pulse should be compensated only at the 50% mark. people use the pitch shift caused by the other pulse widths to intentionally cause pith shifts.

image-et-son, 2022-08-15:
> I think polymod pulse should be compensated only at the about 50% mark.
> 
> People use the pitch shift caused by the other pulse widths (not about 50%) to intentionally cause pitch shifts (there is a really good prophet 6 video on you tube that shows this). For example, using PWM and polymod you can get some interesting patches that have some FM-ish pitch movement.

I do that too. My idea was not to compensate for the drift in OSC A frequency completely anyway but only to eliminate the voice-to-voice differences in that response. Exponential FM always produces an effective offset, even if the incoming signal has no DC component at all. Theoretically is possible to compensate for this by a DC offset (this is exactly what correcting the freq A is doing) but it is a complex formula and to make that work sufficiently precisely will not be possible in the firmware. So it is better to accept that you need to finely adjust poly-mod, waveforms and frequency A to make it harmonic and then concentrate on making this consistent for the 6 voices. That is the plan.

I found a nice publication on FM. See attachment:
[DAFx2020_paper_61.pdf](https://github.com/image-et-son/p600fw/files/9335757/DAFx2020_paper_61.pdf)

image-et-son, 2022-08-15:
@matrix12x : maybe as a check in your modified unit with DC (largely) removed. I expect that you still get a shift in osc A frequency as you turn up the poly-mod amount due to the nature of exponential FM. But, for a given poly-mod strength, how do the osc B waveform selectors affect the shift? Do you get a change when you select only TRI or only SAW vs. when you select both? Is that "additive", i.e. sum of required correction amounts for either is equal to the required corection amount when both are active?

matrix12x, 2022-08-15:
test results: (after 30 min of warm up and pressing tune)

VCOA at C0 saw (mix 100% VCOA zero% VCOB)
VCOB at saw at polymod VCOB full 
Pitch of VCO A drifts to +15 cents for all 6 voices, dead on same for all 6.

VCOA at C0 saw
VCOB at tri at polymod VCOB full 
Pitch of VCO A drifts to +17 cents for all 6 voices, dead on same for all 6.

VCOA at C0 Saw
VCOB at square (50%) at polymod VCOB full 
Pitch of VCO A drifts to +53 cents  for VCO 1
Pitch of VCO A drifts to +53 cents  for VCO 2
Pitch of VCO A drifts to +50 cents  for VCO 3
Pitch of VCO A drifts to +40 cents  for VCO 4
Pitch of VCO A drifts to +43 cents  for VCO 5
Pitch of VCO A drifts to +40 cents  for VCO 6

VCOA at C0 TRI
VCOB at square (50%) at polymod VCOB full 
Pitch of VCO A drifts to +53 cents  for VCO 1
Pitch of VCO A drifts to +53 cents  for VCO 2
Pitch of VCO A drifts to +53 cents  for VCO 3
Pitch of VCO A drifts to +40 cents  for VCO 4
Pitch of VCO A drifts to +43 cents  for VCO 5
Pitch of VCO A drifts to +40 cents  for VCO 6

VCOA at C0 saw (mix 100% VCOA zero% VCOB)
VCOB at saw and TRI at polymod VCOB full 
Pitch of VCO A drifts to +30 cents for all 6 voices, almost same for all 6.

VCOA at C0 TRI (mix 100% VCOA zero% VCOB)
VCOB at saw and TRI at polymod VCOB full 
Pitch of VCO A drifts to +27 cents for all 6 voices, almost same for all 6.

all numbers are approximate.

on all waveform setting everything is fully in tune up to a setting of 44 on polymod VCOB

VCOB was set to an arbitrary frequency of 47 for all testing. I wanted something a little off of a harmonic.

matrix12x, 2022-08-15:
@image-et-son can you explain how you measured the 9.3:
"However, there is an offset in poly-mod amount of about 9.300 below which there is no effect. If that threshold is voice dependent hasn't been analyzed yet."

Were you looking at a scope or by ear or something else?

image-et-son, 2022-08-15:
> @image-et-son can you explain how you measured the 9.3: "However, there is an offset in poly-mod amount of about 9.300 below which there is no effect. If that threshold is voice dependent hasn't been analyzed yet."
> 
> Were you looking at a scope or by ear or something else?
> 
> If it is by scope, my P600 does not seem to have anything happen until a poly mod VCOB amount of about 22 or so.

Hi, one thing first: sorry for using the German notation in which the dot ist a thousand separator and the colon is the decimal separator. With "9.300" I meant 9 thousand three hundred.

My measurement: I started at poyl-mod zero and tuned both oscillators to the same pitch. I then turned up poly-mod in steps of 10 and adjusted (lowered) freq A to match the pitch again. I did this for various osc B shapes and all 6 voices. You see one such plot below (I actually saved every such patch and extracted the 1 bit value from the SysEx using the python tool). 

[image]

The relations are pretty much linear. If I use only the data points from poly-mod 10000 and above and do a linear regression all lines pretty much intersect at poly-mod 9300. The calibration is basically about determining the slope as a function of osc B wave shape.

matrix12x, 2022-08-15:
Oh. ok, so, for my synth, on Saw for VCOA and Saw for VCOB, the pitch for VCO A stayed at C0 (or whatever note I played) until I had a poly mod VCO B amount of around 44 or so. This was consistent all the way up the keyboard. That is, there was no need to re-tune VCOA back to C0 until around 44 or so. The value seems to be 28,864 (28.864) when I look at the saved patch using your command line sysex patch tool. (best tool ever)

Which seems considerably higher than on your unit(s) (about 28K vs about 9K). I wonder if this is the effect of the DC blocking cap (it should be).

How many cents is your VCOA pulled by (detuned) at max poly mod VCO B?

image-et-son, 2022-08-15:
> Oh. ok, so, for my synth, on Saw for VCOA and Saw for VCOB, the pitch for VCO A stayed at C0 (or whatever note I played) until I had a poly mod VCO B amount of around 44 or so. This was consistent [...]

I don't know exactly. You can see in the plot above the for TRI+SAW+PULSE(50) a starting value of freq A of 24600 needs to be compensated to 14000, so pretty strong. 

Between 20 and 44 of poly-mod, do you hear an FM onset, I mean does it do recognizable FM and you simply don't get the pitch shift until 44? That would sound odd, even with the caps.

matrix12x, 2022-08-15:
Yes, between 20 and 40 I get an FM sound. Very faint sound at 20, but I see it on the spectrum analyzer, increasingly louder as I approach about 44. Correct, the pitch shift stays under +1 cent until about 44 at which point it goes to about +2 cents. It is well over 50 before it goes to about +5 cents out of tune.

The main reason I did the mod (originally) was because of how much DC was on the line coming from VCOB when any waveform was enabled. There was a noticeable reduction in offset with the addition of the cap, and based on this data we are discussin, that confirms it. It looks like, based on the data above, you are retuning by almost entire notes at max (unless my math is off).

On my unit if I go full polymod VCOB amount, with VCOB on pulse, if I tune the PW to a narrow pulse, I can get VCOA back in tune easy just by changing the PW without touching the VCOA frequency.

matrix12x, 2022-08-15:
I wonder if some of the DC offset you get from the non-pulse waveforms comes from this (page 2-16 of the service manual):

[image]

Which is addressed by lines 631-645 of Synth.C

Along those lines, do you get pitch shift from no waveforms selected (all 3 off on VCO B), when you turn the bug on and off?

image-et-son, 2022-08-16:
Good thinking. I do get a pitch shift from Pulse Bleed Bug = ON with no waveforms selected, but absolutely no movement when  Pulse Bleed Bug = OFF. All the measurments I did were with Bug = OFF. 

Given the behaviour of your instrument with th DC component removed, I now conclude that most of the pitch movement comes from DC components. I would now expect that you have only the inherent exponential FM pitch response which is like Figure 3 in the paper I attached, i.e. not linear, and this is why you would notice it only much "later" in poly-mod strength. Certainly the better design. 

It probably also means, that it will not be viable to correct for voice-to-voice variations with your mod or at least it would be different. Question: when you use strong poly-mod (>90) and compensate for the drift, how strong is the variation in the voices? Your measurements show some a variation 1-6. Can you still play chords nicely with strong poly-mod?

matrix12x, 2022-08-16:
I'll do more formal testing tonight, but from my recollection the small spread across voices was not too bad with chords. But I usually only play triads. I'll try it out with 4 and 5 note chords.

matrix12x, 2022-08-16:
ok, tested. Using all waveforms on (for VCOB), retuning VCOA so it is in key (i.e., pressing C1=C1) and VCOB so it is in tune with VCOA. 

I held 4, 5 and 6 note chords and they sound good. 

I did the same for VCOB=Saw only. Sounds fine playing up to 6 note chords.

image-et-son, 2022-08-17:
Hey, Stan Jungleib is still around an active, you could ask him :-) Seriously, he might actually respond and might know a lot more than this, too.

https://www.jungleib.com/

matrix12x, 2022-08-17:
Thanks. Just sent him an email.

image-et-son, 2022-08-18:
> Thanks. Just sent him an email.

Exciting! I'm really curious if he responds and what he has to say.

matrix12x, 2022-08-18:
I heard back from Stan. he is super nice. He indicated that the polymod circuit may not have had a cap "because the LFO needed to go fully to zero." He did indicate he does not remember any detail from that period. I personally remember nothing from 1982.

I would like to note that the P600 uses the same circuit for the CEM 3340 (expo FM with no DC blocking cap) as the Prophet 5 and the T8 (and a few Roland synths from the period). 
https://electricdruid.net/cem3340-vco-voltage-controlled-oscillator-designs/
It is very likely that Dave smith just copied his own Prophet 5 design as opposed to redesigning the circuit from scratch, so whatever design choices or errors were in the Prophet 5, would have likely carried over. What I can say is that I have no issues with very narrow pulse widths when using the pulse for FM. I wonder if the intent was maybe to intentionally allow the DC offset so that when you use a pulse that is being PWM'ed you can make a sound where you cause the pitch to intentionally jump up and down.

image-et-son, 2022-08-19:
Cool, I saw how he presents himself on his pages and guessed that he might be approachable, as you confirm.

I am not sure I understand the point with the LFO. The LFO is not hardware imlemented. I don't see the connection!? 

The link is very useful, thanks.

matrix12x, 2022-08-19:
yeah, I was not sure I got the connection either, but the LFO in the Prophet 5 was hardware implemented, so if Dave was reusing the VCO from the Prophet 5, any design choices would have likely carried over.

image-et-son, 2022-08-19:
OK, I see - at least they knew it was going to work. I will check the P5 schematics also. I am not sure if I'd like to share these kind of stories - it somehow takes magic out of things, where you assume that a lot of thought has gone into the design at every stage. I guess the market was moving super fast at the time, so there was no time for iterations.

matrix12x, 2022-08-19:
I know they were working on the T-8 at the same time. I had heard a story once that Dave took about two weeks to design the P600. I was an engineer and I can imagine doing this especially when you use common portions from your other products. 

In the end, a lot did go into the design, even if it was common across a few synths. It's like leveraging blocks of common code that you already pressure tested.

re: sharing these types of stories. Let me know if you want me to remove these last two posts of mine.

image-et-son, 2022-08-19:
> re: sharing these typesot of stories. Let me know if you want me to remove these last two posts of mine.

No, I meant more like when explaining things to a broader audience (in a blog or so) this is not exactly the stuff which legends are made of. The discussion here is only for the hard boiled anyway.