image-et-son/p600fw #189: Random Arp issue

matrix12x, 2022-08-08:
version rc1 v2022 (release candidate 1)
 
Press arp assign 2 times so the led blinks
any single Key gives you C0, does not matter which key you press (e.g., A1 plays C0).
play a chord and the first note starts on C0, but then goes to all the correct notes in a chord.

image-et-son, 2022-08-09:
I think this is the same as #183, correct?

It is fixed in the release :-)

matrix12x, 2022-08-09:
sadly, in the prior release I got a C0 on random Arp the first time I hit any note, but it cleared up on releasing the first note and hitting any note. for example I press arp assign 2 times so the led blinks and I press D3 (or anything else) ... C0 plays, I release D3 and press D3, D3 plays.

With the new release now it is as described above in the first post.

image-et-son, 2022-08-09:
With "new release" you mean the v2022 I tagged yesterday, right? 

I think I solved those issues (there was a clear bug in the implementation which I removed). To be precise: when no note was registered, the internal current index was set to -1. When playing the first note this doesn't matter for up/down and assign because the first thing is to increment the index. For the single note random arp you actually stay on the index, so the (max-1)th notes is played, which is zero (=C0). It is resolved when a second note is registered. That is exactly what you describe. I think fixed that. And when I try it, it plays the correct note every time rght from the start.

matrix12x, 2022-08-09:
correct version rc1 v2022 (release candidate 1)
I re-flashed and still have the same issue I described above

image-et-son, 2022-08-10:
Ah, OK, I have already the created "the "release. Can you please try this: https://github.com/image-et-son/p600fw/releases/tag/v2022

I hope I don't need to change this...

matrix12x, 2022-08-15:
I just tried the version you have linked above and I do not get the random arp issue I noted above. 
ok to close issue.

image-et-son, 2022-08-16:
ok, good