image-et-son/p600fw #190: About 2 glitches occur between poly-MOD OSC B values 5 and 6.

Jpro600k, 2022-08-15:
About 2 glitches occur between poly-MOD OSC B values 5 and 6.

[URL=https://drive.google.com/file/d/1-F-9ql7GAw_jn_n-CAJ7etrqx7c4J-6K/view?usp=sharing]20220815_A glitch occurs around the poly-MOD OSC B.zip[/URL]

Jpro600k, 2022-08-15:
I was able to identify the value.
50 and 54
A value of 50 shows 00
A value of 54 indicates a random value
Glitch starts when you press a key
https://drive.google.com/file/d/19eGKnEjkErxeRNq-GYKck-P0DJ8cijKX/view?usp=sharing
The range of values is an important area when creating sounds. can you fix it?

Jpro600k, 2022-09-07:
Is it a problem that takes a lot of time to fix? 
Quite inconvenient because I use poly-MOD OSC B so often.

image-et-son, 2022-09-08:
Hi, it is one of the issue which are hard to pin down, because it is unit specific. On my second unit I have minor glitches on "Mix" but only when the unit is cold. The brute force method would be to extend the wait time for the mutliplexer to "settle", but that would come at a high cost - it would slow down everything. I also cannot do it specifically for the "Poly-Mod Amount" because for someone else the problem would be something else.

Can I ask you about your observation?:
- Do you experience glitches on any other knob, be it only occasionally?
- Does the behaviour change when the instrument is warmed up?
- What's your 418 multiplexer, e.g. the last one of the four half-way under the keyboard?

Note that it could also be that you have some dirt or dust in the pot. If it comes to it, I may send you a testing version with more mux settle time, just to probe if the problem is related to software speed vs. hardware inertia.

matrix12x, 2022-09-08:
@Jpro600k if you use MIDI cc to change polymod do you get the same glitches at the same values?

also I could not reproduce this issue.

Jpro600k, 2022-09-08:
> @Jpro600k if you use MIDI cc to change polymod do you get the same glitches at the same values?
> 
> also I could not reproduce this issue.

Thanks for the suggestions. 
Operations from Prophet 600 Black and Red Ctrlr Panel_1_0  were glitch-free.

Jpro600k, 2022-09-08:
> Hi, it is one of the issue which are hard to pin down, because it is unit specific. On my second unit I have minor glitches on "Mix" but only when the unit is cold. The brute force method would be [...]

Hi, imogen

All other knobs were fine.
U418 is MC74HC4051N.
This chip is not slow and shouldn't have any speed issues. Could this chip be the problem?

Thanks for the suggestion about the dust problem.
I think it was a great suggestion. I forgot that this is old equipment.
I turned the knob 50 times with such speed that it seemed like it would break. and then checked again. 
??? The glitch may have cleared !
I will check the situation for a while.

image-et-son, 2022-09-08:
ok, glad it's not a bug :-)

closed 2022-09-08 by image-et-son