image-et-son/p600fw #195: Voice deactivation wrong in unison mode

image-et-son, 2022-09-08:
The deactivation of voices is not properly taken into account in unison mode, i.e. deactivated voices play anyway. Also, when no voice is activated, playing any key in unison crashes the firmware completely.

jmechnich, 2022-10-08:
Not sure if this is related (or even normal / expected behaviour?):
In unison mode with priority last, if one plays and holds in series C-D-E-F-G and starts releasing the keys from the top down, the sounding pitches will be:

```
PRESSED   - SOUNDS
C         - C
C-D       - D
C-D-E     - E
C-D-E-F   - F
C-D-E-F-G - G
C-D-E-F   - C
C-D-E     - D
C-D       - C
C         - C
```

My naive expectation was that the 'history' of key presses is kept to a certain point but after looking at the code it does not seem to be the case. Anyways, it seems a bit strange that it will keep switching between C and D after each released key (try more than five keys :smile:).

image-et-son, 2022-10-08:
Hi! No, that's different issue, but I have already created a ticket for it: #106. But well spotted!

I think it would be proper to take the timestamp into account, but I am not sure if it is worth the effort (or rather: the additional firmware storage space that would cost).