image-et-son/p600fw #215: Do i have a dud?!

TheRaydevil, 2023-02-04:
Hey everyone, My nice GliGli upgrade is unfortunately not going as planned... Bought a Teensy++2.0 from Tinkesphere, Prepped it as per instructions, loaded the 2.0V hex file, installed it... power on and... Nothing. Took it out again and made sure to check everything top to bottom, over and over. Everything in order. re-loaded the file a couple of time... still nothing. Finally Decided to try loading it with Imogens version 2.1, Re-installed it, power on and now i have the tuning sequence starting but crashes right after going thru the filters leaving the number 6 on the display. It just freezes like that. Tried over and over, even the GliGlis 2.0V one last time. Nothing.

I can only get it to "power up" with Imogens version but then of course it freezes. Any other version will do nothing. If i put the original Z80 back in everything works normal. I sort of lost now... Any ideas?

defcon8, 2023-03-24:
I have the exact same issue.. any help?

matrix12x, 2023-03-24:
They are likely clone Teensy boards, and will not work.  
There is no current fix for using clones.

defcon8, 2023-03-24:
@matrix12x:

Any way to identify that the board is a clone? I did notice that my boards indicate 3.3V on the bottom where the one in the instruction pictures has 3V. 

Images of my board;
https://ibb.co/Qc6hjPp
https://ibb.co/CHm5NjR

matrix12x, 2023-03-24:
I'm not sure. There has been some discussion on gearspace on the P600 thread.
If you dive through the thread long enough, there is even technical conversation as to why the clones don't work, and how to possibly identify them.
https://gearspace.com/board/electronic-music-instruments-and-electronic-music-production/845271-prophet-600-users-have-upgrade-138.html